atc1441/ATC_TLSR_Paper #1: Firmware doesn't work on Stellar-MN@ E31HA 401011

dustybee, 2021-12-03:
Hi,

The included firmware doesn't display anything and doesn't advertise BLE. I have confirmed the firmware does run by toggling the red LED in main.c. 

It does seem my MN@ display's have slightly different product codes than yours but have the same PCB text.
I have been able to get working firmware on the display by rewriting the epd_driver_demo in the TLSR8359 SDK to support the display.

The displaying part might be because of a different EPD display, but that doesn't explain the tag not advertising BLE because it is a TLSR8359 and the same PCB.

Do you have any idea on what might be wrong?

Also, I had to change line 4 of the makefile:
```TC32_COMPILER_PATH := tc32_windows\\bin\\``` to ```TC32_COMPILER_PATH := tc32_windows/bin/```
because of error ```/usr/bin/sh: tc32_windowsbintc32-elf-gcc: command not found```

running Python 3.9.6 on Windows 10

atc1441, 2021-12-03:
Thanks for that. 

The not working BLE could be a different revision of the Chip. As this is not an official "feature" everything can happen.

Also everything is very early stage and not "useable" as mentioned. 

A check of the chip revision could be added to see if there any different or something

Jeroen207, 2022-02-02:
I found an interesting document that states: 

> - **Model No:** Stellar-M3N@, Stellar-M3YN@, Stellar-MN@
> - **Diff:** All above models are identical in the same PCB layout, interior structure and electrical circuits. The differences are when product at work of EPD Screen color and model name for commercial purpose. This report performs the model Stellar-M3N@.

So it seem that they use the same microcontroller, but the screens can indeed be different. However, it is still strange that it doesn't advertise BLE. The one I use is a: Stellar-M3N@.

Source: 
[https://www.tele.soumu.go.jp/giteki/SearchServlet2?PageID=jt01&ATF=33397004](url)

dustybee, 2022-02-02:
Hi,

The BLE ended up working with a later version of the custom firmware.

The only thing that still isn't working is getting anything to display. Even after changing the display EPD driver in the ATC_TLSR_Paper firmware to one which I know works in an empty project, it still won't display anything, but that will be something for later.

@Jeroen207,
> The one I use is a: Stellar-M3N@.

I have also tried the firmware on a Stellar-M3N@ but didn't have any luck getting it to display anything.

Jeroen207, 2022-02-02:
That is weird that BLE works with the newer version, I have to try again myself. I also did some changes with the display driver myself, but I was too focused on the BLE. 

I have to ask my local grocery store (AH) if I can get a working one, since I'm not quite certain if the ones I have works for a 100%. They tossed 2 of them out. (I only figured out that the battery was empty).

dustybee, 2022-02-02:
Hi @Jeroen207,

To test of your display's are working, try uploading the attached bin file via [the flasher](https://atc1441.github.io/ATC_TLSR_Paper_UART_Flasher.html). once it's done uploading I always have to click "Soft reset MCU". 
[OUTPUT.zip](https://github.com/atc1441/ATC_TLSR_Paper/files/7989028/OUTPUT.zip)

dustybee, 2022-02-02:
@Jeroen207

I downloaded the TLSR83589 SDK and modified the EPD_DRIVER_DEMO.  The only problem I am having is extracting just the EPD_DRIVER_DEMO project from the SDK for easy modification. It's a very tedious process to modify in VSCode, compile the project in eclipse and upload the firmware via the web flasher.

I'll upload the source code to my GitHub. I didn't do this at first because I am not proud of my hacky code 😅.

artelse, 2022-06-16:
I am having the same M3N@ display. Aaron's code loads, but after manual soft reset shows the intended info with a lot of random flashing and noise. Eventually it 'settles down' and flashes the green LED.
The OUTPUT.bin file loads as well, but no screen refresh. Does show the red LED though. Any ideas?

artelse, 2022-06-16:
Display model MN@ works just fine; update via UART and then OTA!