atc1441/ATC_TLSR_Paper #4: Barcode and unique identifier of each ESL tag

martinberlin, 2022-01-15:
Hi @atc1441 ,

Would it be possible to have a:

    const char * serial = "517714";

I was just analyzing the Barcode and it seems to me an unique identifier:
[image]
Which has this number in my device:
[image]

Then if we build this firmware using that constant, we could get a bluetooth name like:
**ESL_517714**

My idea would be that we could scan this Barcode with a custom App and then we could automatically guess what is the device name. This can help to associate a device faster instead of browsing the hundred bluetooth around this days. But is just an idea, would be cool to have an unique serial and I just wanted to know where is the part of the code that generates the ESL_number.
For this device the name I get is: ESL_545778

On another question, I'm designing a small 3D-printable tiny placement, in case you want to place this vertically in your table with a nice inclination so you can see it more clearly. Will share it in Thingiverse once done and tested!

atc1441, 2022-01-15:
Currently the name of the tag is based on its MAC address.

The barcode could be the hanshow serial number and generated on the manufacturing line. It will be saved somewhere in flash but after erasing it and writing the custom firmware onto it it will be gone.
So it would be needed to put it in manualy

martinberlin, 2022-01-15:
Thanks!  Closing here since I think will be pretty easy to change this if I want to.
I will try to help testing Larry app while there is still some permissions issue and I cannot install it on any of my Android devices.

closed 2022-01-15 by martinberlin