atc1441/ATC_TLSR_Paper #5: Latest commit breaks previous working upload image using BLE

martinberlin, 2022-01-20:
Last working stable commit is this one:

commit ff957594fc5747ecaf2c789c4abadef7b54c6be9
Author: atc1441 <atc1441@googlemail.com>
Date:   Sat Jan 15 17:24:03 2022 +0100

    Final CRC check working

Sorry but I would really like to recommend to make a feature branch where others can test and only after testing merge in master. 
Otherwise it's not so cool to break existing working behavior and just push it to main branch without letting at least other developers test the new feature.

atc1441, 2022-01-20:
The crc check is only for ota firmware update.
Image upload still works as usual and fine.

Please note where it fails. Does it not upload at all? What is happening

Sorry to say it this way but how i handle the pushes is the way i do. I do not like branches.

martinberlin, 2022-01-20:
It does upload the image but it refreshes again with Bart. I tried to revert it to commit ff95759 and build it again locally, flashed it. And then it works as before. Tried with 2 sites, with CALE and with your BLE upload. 
Will try again, but there is something, maybe the binary was broken?   Will report after testing it a second time.

martinberlin, 2022-01-20:
Alright pulled latest changes:#
$ git pull
Updating ff95759..60e7e30
Fast-forward
 Firmware/src/app.c             | 4 +++-
 Firmware/src/epd.c             | 5 ++---
 Firmware/src/epd_ble_service.c | 7 ++++++-
 3 files changed, 11 insertions(+), 5 deletions(-)

Built again locally and now it works. I see that the bin is modified, maybe that was it?
That I just had to build it?   It's Ok no problem I understand I also don't like branches. 

But then unless it's very well tested it will inevitable have master versions that break at some point backwards compatibility. At least it would be cool to have one extra branch develop and test there, white others can help testing as well, before making a new release.

martinberlin, 2022-01-20:
Alright, it seems that was something about "soft resetting the MCU" before sending the image.
Also can be that the Firmware sometimes does not arrive 100% OK and you just need to flash it again. Still, even if you don't like it, I think my recommendation is valid.

closed 2022-01-20 by martinberlin