atc1441/ATC_TLSR_Paper #7: Upload failed

defconhaya, 2022-02-02:
Hi,
I've tried to upload the firmware but nothing changed. I've got no error messages but the image doesn't changed and the display did not even blink.  I've used 2 different USB2TTL devices: based on cp2102 and ch340 chips. Is there any special thing that I must consider when making the upload ?
Thanks.

atc1441, 2022-02-02:
While i can not help to find whats happening wrong i can give some info.

The flasher does not know if any commands/flashing works so it will continiue without error.

You connected sws and reset?

defconhaya, 2022-02-02:
Sure, both SWS -> TX and reste->RST are connected, maybe the MCU is not in download mode. What usb2ttl adapted did you used with success ?

dustybee, 2022-02-02:
Hi,
You mentioned you connected:
>SWS    ->   TX
>NRST  ->    RST

Have you also connected GND -> GND?
Also, make sure you connect NRST -> RTS on the TTL converter. RST is an input on the converter.

defconhaya, 2022-02-02:
I finally managed to flash one board, but it was the one I've removed the panel. I've seen it when I scanned for BLE, it has something like ESL_ name, but it is the bare board. 
Thanks for your time, and keep up the good work 👍

closed 2022-02-02 by defconhaya