atc1441/ATC_TLSR_Paper #9: BLE not working on Stellar-MN@ E31H 401013

leo-lb, 2022-02-13:
Hello!

I tried flashing various versions of this firmware with no luck getting BLE working, flashing works as flashing some test firmware that makes LED flash works.

How exactly is BLE working with TLSR8359? How can we test it's presence in hardware?

Thank you

Jeroen207, 2022-02-13:
As mentioned multiple times by @atc1441 

The TLSR8359 chips officially are not compatible with BLE.
The only way to make them work is to hack an SDK from another chip. This is why it works.

But still we have not figured out fully why it don’t work, it could be another reversion of the TLSR microcontroller. But in the end, it is a shitty Chinese microcontroller with a terrible SDK library & IDE.

artelse, 2022-06-16:
The MN@ version I have works fine with BLE.