atc1441/ATC_TLSR_Paper #11: Hanshow Stellar-XLN@

artelse, 2022-06-06:
I have a Stellar-XLN@ model which is a larger display of 84.22*63.02 mm and a battery life of 5 years. I have no idea what CPU is on board. I opened up the battery departments, and it reveals a test connector with following labels:
```
| O | O | VCC  | GND  |
| O | O | TMS  | NRST |
| O | O | RXD  | TO?  |
| O | O | TEST | TXD  |
```
The labels on the second row are hidden behind the plastic casing, but I think they are correct, except TO?
The other Stellar tags have a SWS connection that I don't see here. Any idea what TMS and TEST refer to, is one a SWS pin?
Thanks for your great work on the Hanshow ESLs.

atc1441, 2022-06-07:
Hey, the XL uses a TI CC2640 SOC

atc1441, 2022-06-07:
Working on it more or less :) 
But its nothing with priority as the soc is so rare and needs a bit of care to get into

artelse, 2022-06-07:
Is the CC26xx not used a lot in wearables?

artelse, 2022-06-11:
Have you been able to hook it up to comms? Have you traced the debug connector to the pin layout? Would love to have that info, otherwise need to break open the case.

sengelsman, 2022-10-01:
I've found exactly the same connector inside a Stellar-XL3N@

```
| O | O | VCC  | GND  |
| O | O | TMS  | NRST |
| O | O | RXD  | TO?  |
| O | O | TEST | TXD  |
```

Should i try to obtain a CC-debugger if i want to do anything with this SoC in the future?

atc1441, 2022-10-01:
> I've found exactly the same connector inside a Stellar-XL3N@
> 
> ```
> | O | O | VCC  | GND  |
> | O | O | TMS  | NRST |
> | O | O | RXD  | TO?  |
> | O | O | TEST | TXD  |
> ```
> 
> Should i try to obtain a CC-debugger if i want to do anything with this SoC in the future?

That one is not supported by cc debugger.

Maybe there will be a custom Firmware in the future. But a loong future

MoJo2600, 2022-10-06:
It seems like the M3@ has the same connector inside. The version printed on the board is HS_EL5102_0_V2-4. Do you have some hints on how to get started on this Microcontroller?

MoJo2600, 2022-10-06:
I think I am wrong... I opened one of my tags and there are two chips "MSP430 G2755" and "A7106". I don't think they use the CC2640 SOC at all.

gabimal, 2024-12-22:
> That one is not supported by cc debugger.
>
> Maybe there will be a custom Firmware in the future. But a loong future

I open my Stellar XL3N ESL and found these:
[image]
[image]
[image]
[image]
[image]

I didn't find any support for 4.2 inch screen size so I flashed the fw for M3N one, just for fun. Now I have a funny screen with white, black and red pixels :)
Any news or work in progress for this screen?
Thank you in advance!