atc1441/ATC_TLSR_Paper #12: M3N@ / BWR213 won't detect right version?

artelse, 2022-06-18:
As reported in another thread here, the M3N@ version doesn't refresh its display properly. I flashed it thru UART and can now update it OTA. But the problem with the garbled display might come from not detecting the right BWR213 version. It displays it as BW213 which is not correct. Forcing the model version OTA, has no effect.

atc1441, 2022-06-18:
Sounds like its another version again, but since i dont have it i can not support right now

artelse, 2022-06-18:
I assumed it was the BWR213 because when it refreshes it has the additional red color and it is on the list of supported devices. 
Is there a certain order when `Force model` is applied? I tried a few to no avail.
What does the `0055` command do, cannot find it in the `cmd_parser `.

atc1441, 2022-06-18:
Please share the full name of your label here

artelse, 2022-06-18:
Stellar-M3N@ E31HA 
I see you check the right model# throughout, so that should work.
On this line in `epd.c` it should report the right model on the display while it doesn't do that.
https://github.com/atc1441/ATC_TLSR_Paper/blob/d99ad9f71b3589c529571569c37d7a26270cb577/Firmware/src/epd.c#L270
Still wonder what `cmd` 0055 does..

atc1441, 2022-06-18:
Cmd 0055 https://github.com/atc1441/ATC_TLSR_Paper/blob/main/Firmware/src/epd_ble_service.c#L32

artelse, 2022-06-18:
Don't know if this helps, but the PCB shows `HS_EL5102_1M_62_04` which is the same as on the picture with the uart connections.

Jeroen207, 2022-12-02:
I do have the exact same one lying here with the same problem. 

> Don't know if this helps, but the PCB shows `HS_EL5102_1M_62_04` which is the same as on the picture with the uart connections.

Did you got them from a "specific grocery store"? I do have the same issue with same M3N@.

artelse, 2022-12-02:
> Did you got them from a "specific grocery store"? I do have the same issue with same M3N@.

Yes, got it with a Bonus offering ;)

Jeroen207, 2022-12-02:
Gotya, @atc1441 if you like I can send you one to "experiment" with it.

Jeroen207, 2022-12-02:
> > Did you got them from a "specific grocery store"? I do have the same issue with same M3N@.
> 
> Yes, got it with a Bonus offering ;)

I have used custom made code from [dustyBee]( https://github.com/atc1441/ATC_TLSR_Paper/issues/1#issuecomment-1028213718nv), I experimented with it long time ago, but I was travelling around the world and just left it alone for a while.

He used the SSD1680_EPD display driver which worked great for BWR colours. I understand the confusion because
"specific grocery store" only use black and white while they have an option to use red as well.