atc1441/ATC_TLSR_Paper #15: How to debug project ?

garobcsi, 2022-08-02:
How could i debug this project so that i could add modifications to the project.

garobcsi, 2022-08-03:
I have no idea.

garobcsi, 2022-08-04:
@atc1441 any help would be greatly appreciated.

atc1441, 2022-08-04:
Hey, in general you can just compile it and use the UART TXD pin on the back to get and write your own debug outputs to make it easy to debug better :)

garobcsi, 2022-08-04:
So if i use the uart.c and i write something out to it i could just read it w/ uart to usb ?

garobcsi, 2022-08-04:
Also a usb to ttl would work for the flashing ?

atc1441, 2022-08-04:
yes,
you can use the normal printf like here:
https://github.com/atc1441/ATC_TLSR_Paper/blob/main/Firmware/src/main.c#L46

so far every uart converter worked in the end

garobcsi, 2022-08-04:
So i innit the uart and i could printf and use the uart on the back to get message. Thx for the help.

atc1441, 2022-08-04:
Yes, but uart is already inited :)

garobcsi, 2022-08-13:
Idk why but printf didn't work but puts void worked.

closed 2024-01-11 by garobcsi