atc1441/ATC_TLSR_Paper #16: Paper_Image_Upload

garobcsi, 2022-08-12:
Hi. I successfully flashed the e paper label and i saw u have a image uploader but there is no clear way (for me) to make my own image could you reccomed a program that could be useful for this (i saw one but i cant find it) or any other solution that could help.

garobcsi, 2022-08-12:
i tryed everything with the img flasher nothing works even the test img is broken

atc1441, 2022-08-12:
The best is to add a crc check for the upload and retransmit on failed ones.
It could be a problem that your device is sending the data too fast

garobcsi, 2022-08-12:
And how don add a crc check ?

atc1441, 2022-08-12:
Best position would be in this call https://github.com/atc1441/ATC_TLSR_Paper/blob/main/Firmware/src/epd_ble_service.c#L38
The crc value needs to be added on the sending page as well of course

garobcsi, 2022-08-12:
On the side i send a cmd 0x00 ?

garobcsi, 2022-08-12:
Site*

garobcsi, 2022-08-12:
so i enabled the slower sending of the img and its clearly was slower but it still put out a broken img

garobcsi, 2022-08-13:
So i flashed my second device and the same problem occurred.

Bogoul, 2022-08-29:
Hi
Same for me all the pictures are like the one above.

Thank for your work.

garobcsi, 2022-08-29:
So this is a bug... great. Thx for notifying me.

sengelsman, 2022-11-09:
I've been trying to get the example image to display correctly on some BWR213 250x122 displays.
Finally made it work by changing the hardcoded `epd_height` and `epd_width` in `Firmware/src/epd.h` to the resolution of my display.
<edp.h> line 3
```
#define epd_height 128
#define epd_width 250
```

garobcsi, 2022-11-09:
Oh thx

garobcsi, 2022-11-09:
works perfectly Ty 
[image]

sengelsman, 2023-02-09:
> I've been trying to get the example image to display correctly on some BWR213 250x122 displays. Finally made it work by changing the hardcoded `epd_height` and `epd_width` in `Firmware/src/epd.h` to the resolution of my display. <edp.h> line 3
> 
> ```c
> #define epd_height 128
> #define epd_width 250
> ```

Today when flashing the firmware with this change to an epd_bwr_154 (or very close relative), the device would hang after forcing it to epd_bwr_154 and only a battery removal would make it available over the air again. After changing `epd.h` values back to the default 200x200 and re-flashing i could force the device to epd_bwr_154 and it would be stable afterwards.

Ward11, 2023-06-19:
Hi, how do you exactly send an image to the deveice? I have no Idea how to use the image uploader, are there any tools available to create the right code to send to the ESL?

garobcsi, 2023-06-22:
@Ward11 
Img to Hex

[Hex example](https://github.com/garobcsi/ATC_TLSR_Paper/commit/210831c778bbb150bd09c270288aa88c4ed9bc4d)

UserXYZ, 2024-01-02:
This is pointless answer as just saying Img to Hex leads to nothing....
There are quite a few web sites where you can do conversion img2hex and hex2img, and in whatever format you give them the input image, you get useless bunch of hex bytes that represent nothing when you upload them to the tag.

To make matters worse, some of the repositories built on original do have some of the tools to manage images: stellar-L3N-etag for example has even whole web page (in Chinese though) that manages images greatly, but the code is not compatible with ATC1441 and seems to only work on one model, 296, which is not even listed on Hanshow's site.
It also has some python utilities which actually generate nothing as they are trying to import python modules that do not exist.

Mom test image that garobcsi created does work, but he never answers how did he create it - what tool or command was used, what was the input image (format, size etc).

Therefore, the original question - how does one convert own image to proper stream of hex bytes is still valid and important.

garobcsi, 2024-01-02:
https://github.com/garobcsi/ATC_TLSR_Paper

I made my own mofifications to the project. Just flash the firmweare and use the modified web tool. And if you enable image mode the image will stay forever (until you change it).

UserXYZ, 2024-01-02:
Hi garobcsi, I got a copy of your repo and flashed the code to one BWR213 I have.
It gives very similar experience to the original stellar-L3N-etag project, without the Chinese, but it also does not work with smaller BWR154 model which works just fine with original ATC1441 code.

So your version of webtools is quite useful but only for one of the tags I have.
Unfortunatelly it has a bug in image encoding/dithering - red color is replaced with black on the display, even though it is correctly shown and dithered on the web page, and the correct tag type is detected (even fixing it manually in epd.c makes no difference).

Original stellar-L3N-etag repo does not have that problem so the same BMP image I used for the test shows fine.
If I use the canvas to draw few red lines, they are also replaced with black lines in the end, so there has to be some difference between your code and original from stellar-L3N-etag repo that misses the red color.

As the smaller BWR154 is not working with your firmware, I still need to use original ATC1441 firmware for it, and then I am still presented with the same problem: how to convert input image into hex stream that is valid and can be uploaded to the tag?
What tool did you use to create your 'Mom' picture here: https://github.com/garobcsi/ATC_TLSR_Paper/commit/210831c778bbb150bd09c270288aa88c4ed9bc4d

garobcsi, 2024-01-02:
Btw for the red and withe and black issue 
[image]
u can just change the algorithm.
And the red output only works if it using the 296 mode forced. (Some models already use it forced)
- u can find the mode witch useing on the first clock mode (the second clock mode boots by default)

For the Hex convertion i used this on windows: https://www.waveshare.com/wiki/File:Image2Lcd.7z
- Set 250 128 resulution
- export it as .c
- ignore first line bc not as same length as the others
- remove 0x<data>,      (keep the data)
- and you can use it on the image flasher (red image not supported on the old flasher)

On linux i used a tool but i already forgot what i used.

UserXYZ, 2024-01-02:
Thanks for the reply - could you explain what field shows the mode when choosing clock type 1?
For example, my BWR213 shows the top line with this: "ESL_32A95C BWR213 BLE T"
Is it this "BLE T"?

Also, how does one force the mode for 296 model? Is it limitation of the software, firmware or the modules themselves?

garobcsi, 2024-01-02:
BLE T
BLE F

BLE: Bluetooth Low Energy
Ture and False
.
In the web tools the forcing option is possible but not implemented in the web tool.
Also you could modify the firmware to default to 219

Ward11, 2024-01-03:
> https://github.com/garobcsi/ATC_TLSR_Paper
> 
> I made my own modifications to the project. Just flash the firmware and use the modified web tool. And if you enable image mode the image will stay forever (until you change it).

Great, I will try it out :)
Here are some results from fiddling around last summer;
[image]

UserXYZ, 2024-01-11:
Hi @Ward11 , I see in your pictures that you got 3 colors displayed....which device model it is then? 213BWR or 296BWR?
I tried to use stellar-L3N-etag firmware on 213BWR, and then used @garobcsi web tools to control it - it does work, I can even get 3 colors, but the display still resizes itself to the dimensions of 296 so screen is wider than it is for my 213 tags.

So, if you're also having 213BWR, please share how you got the firmware to work properly.

UserXYZ, 2024-01-11:
BTW @garobcsi , I modified these two lines in epd.c to try and force 296 model:
`RAM uint8_t epd_model = 5; // 0 = Undetected, 1 = BW213, 2 = BWR213, 3 = BWR154, 4 = BW213ICE, 5 BWR296`
and
`epd_model = 5`

Did not touch if-then-else parts as they are triggered only if previous is set to auto-detect.

Would that be the correct way?
How about the screen size as well? Is mofidying epd.h also needed?

garobcsi, 2024-01-11:
If you modify the screen size in epd.h that (you should) you also have to modify the web tool too bc it wont be the same size as your screen.
For the autodetect just remove it ?! Idk

garobcsi, 2024-01-11:
Also the web tool is modified for my firmweare.
For other firmweares im not doing comypatibilits

closed 2024-01-11 by garobcsi