atc1441/ATC_TLSR_Paper #17: Hanshow Nebular

lippe-digital, 2022-11-29:
Did someone already got hands on Nebular devices?

They are popping up in stores around here.
They've got BW/Yellow and BW/Red. Also there a ones which are big, like a DinA5 sheet...

lippe-digital, 2022-11-29:
[image]

lippe-digital, 2022-11-29:
[image]

atc1441, 2022-12-01:
Hey. The 3.5" ones are directly compatible and will also do Bluetooth :)

Included the general files today but the display itself still needs the correct custom driving, different resolution and different custom LUT for fast updates

tobiasdroessler, 2022-12-04:
> Hey. The 3.5" ones are directly compatible and will also do Bluetooth :)
> 
> Included the general files today but the display itself still needs the correct custom driving, different resolution and different custom LUT for fast updates

Do these devices have a proper Antenna for bluetooth?

I read some docs about the product,  and there was a NFC interface mentioned.  Is it's just NFC or also WiFi?

tobiasdroessler, 2022-12-19:
@lippe-digital did you try to flash your nebular device successfully?

lippe-digital, 2022-12-19:
No. Didn't have time to get my hands on yet.
But I'm really eager to figure it out.

tobiasdroessler, 2022-12-27:
Now it worked for me..  The soldered battery was may in the meantime quite low on voltage. So I removed it and sourced the nebular with the 3V3 from UART Adapter. The BLE Device is now appearing :). I will look at the source code next days, the display size/orientation not matches for my target.

rck, 2022-12-28:
I guess I was also able to flash the firmware via the web flasher, at least the LED blinked blue during flashing. Aside from that would you expect anything to work? Like should a "soft reset MCU" clear the display or anything?

tobiasdroessler, 2022-12-28:
After intial flashing and a power on reset there is a test routine which writes the whole display black and write for a fee cycles...  And then this appear..  
[image]

atc1441, 2022-12-28:
Nice to see something working out.

The Nebular ones are very very beta or even Alpha, please do not expect it to work usable.

It is just a POC to see the Basics working :)

rck, 2022-12-28:
thanks for the good work. I did not expect anything at this early stage, I'm just wondering why it does not seem to do anything at all in my case. flashing looks good, at least the LED is blue when sending stuff (I did send the unlock flash command), but then it does not show up as BLE device and does not change the display. FWIW this is a Nebular 350-RN. Looks like there are different revisions, my pin out is slightly different from the one shown in this thread. I have a second one that has a display attached (and cleaner soldering :) ). SWS is connected to UART TX, RST to UART RTS, VCC to 3V3, GND to GND. that should be correct, right? I also tried with RST unconnected. RST vs NRST in the other pin outs? anything obviously wrong with the wiring?

[image]

tobiasdroessler, 2022-12-29:
I think your wiring is correct.. Did you unlock and ERASE the flash memory like described in the youtube video?

rck, 2023-01-03:
I did. I don't know why, but it did not work with my flipper zero but did work with a bus pirate. Now I find the device in the BLE app and see output on the display :+1:

WatskeBart, 2023-01-20:
> I did. I don't know why, but it did not work with my flipper zero but did work with a bus pirate. Now I find the device in the BLE app and see output on the display 👍

Had the same issue with a Flipper Zero. My FT232 did the trick though.
Did you manage to upload something to the display?

This is my Hanshow Nebular 213R-N:
[image]
[image]

rck, 2023-01-23:
> Did you manage to upload something to the display?

I only uploaded the initial firmware which then displayed the clock and the the device showed up as BLE device, but that is where I stopped to let others do the actual work :). Refresh did not really work and the display size was wrong, but that is AFAIK currently all expected.

WatskeBart, 2023-01-23:
Got my hands on another Nebular.
[image]

WatskeBart, 2023-01-24:
I have more succes with the Nebular 154. Unfortunately the screen was cracked, but it shows the clock partially. I can't get it to fill the complete display just like @tobiasdroessler
I even tried to changed the value's in epd.h to match the resolution of the Nebular 154
```
#define epd_height 200
#define epd_width 200
```
[image]

WatskeBart, 2023-01-24:
@atc1441 Why is the UART output garbled of a flashed device? Only connected the VCC, GND and TX-to-RX to my UART adapter at 115200bps

tobiasdroessler, 2023-03-21:
I think the reset is needed just before flashing. Did you connected the RTS to the reset pin?

May your adapter not work as intended and the reset is not done. You could also issue a reset manually before click the flash button...  more info may atc can provide.

nethomas1968, 2023-03-28:
I'm using the Hanshow Nebular 350R-N . And, I'm looking at the EPD_BWR_350_detect function. It's trying to write 0xA5 in to 153 bytes of storage locations, and then read them back. I'm presuming this is to test that there are that many bytes of storage available - and from that we can determine the model. 
Well, unfortunately, the read-back is failing at byte number zero. So, it's failing to write 0xA5 in to any of the locations.
Is this a symptom of a bigger problem for the Hanshow Nebular 350R-N ?

atc1441, 2023-03-28:
As mentioned already the Nebular ones are not at all proven to work

atc1441, 2023-03-30:
As usual there is no datasheet available, but the pinout is the same :)

nethomas1968, 2023-03-31:
> As usual there is no datasheet available, but the pinout is the same :)

Actually, it's these pin settings that I'm querying - from main.h :
#define EPD_RESET 	GPIO_PD4
#define EPD_DC 		GPIO_PD7
#define EPD_BUSY 	GPIO_PA1
#define EPD_CS		GPIO_PB4
#define EPD_CLK		GPIO_PB5
#define EPD_MOSI	GPIO_PB6
#define EPD_ENABLE	GPIO_PC5

The code to read the battery power and battery temperature is working fine - but the SPI comms to the display aren't working.
So, I'm wondering if those GPIO pins are different on the Nebular series.

atc1441, 2023-03-31:
The pinout is the same.
But again, there is still work needed to get it fully running.
Sorry for being so short on that but its how it is currently

supergerrit, 2023-06-27:
I managed to get the Nebular 154 fully working. I found that some of the commands were wrong by looking at the datasheet of the display [link](https://www.e-paper-display.com/IL0376Fb93d.pdf). 

I changed the following things:
- Panel setting was incorrect
- Resolution was set incorrect
- Data stop commands were not used after writing the buffer to the device

This is the code i'm currently using:

```C
EPD_init();
    // system power
    EPD_POWER_ON();
    WaitMs(5);
    // Reset the EPD driver IC
    gpio_write(EPD_RESET, 0);
    WaitMs(10);
    gpio_write(EPD_RESET, 1);
    WaitMs(10);

    // Booster soft start
    EPD_WriteCmd(0x06);
    EPD_WriteData(0x17);
    EPD_WriteData(0x17);
    EPD_WriteData(0x17);

    // power on
    EPD_WriteCmd(0x04);

    // check BUSY pin
    EPD_CheckStatus(100);

    // panel setting
    EPD_WriteCmd(0x00);
    EPD_WriteData(0b11001111);

    // resolution setting
    EPD_WriteCmd(0x61);
    EPD_WriteData(0b11001000); // 200 to binary = 11001000
    EPD_WriteData(0x00);
    EPD_WriteData(0b11001000);

    // Vcom and data interval setting
    EPD_WriteCmd(0X50);
    EPD_WriteData(0x97); // 0x97 to binary = 10010111

    // load bw image data to EPD
    EPD_LoadImage_bw(epd_buffer, resolution_w * resolution_h / 8, 0b00010000); // bw
    EPD_WriteCmd(0b00010001);  // Data stop

    // load red image data to EPD
    EPD_LoadImage(epd_buffer, resolution_w * resolution_h / 8, 0x13); // red
    EPD_WriteCmd(0b00010001);  // Data stop

    // trigger display refresh
    EPD_WriteCmd(0b00010010);
```

With the following `EDP_LoadImage()` functions, the data for the red channel needs to be inverted:

```C
_attribute_ram_code_ void EPD_LoadImage(unsigned char *image, int size, uint8_t cmd)
{
    int i;
    EPD_WriteCmd(cmd);
    for (i = 0; i < size; i++)
    {
        EPD_WriteData(~image[i]);
    }
    WaitMs(2);
}

_attribute_ram_code_ void EPD_LoadImage_bw(unsigned char *image, int size, uint8_t cmd)
{
    int i;
    EPD_WriteCmd(cmd);
    for (i = 0; i < size; i++)
    {
        EPD_WriteData(image[i]);
    }
    WaitMs(2);
}
```

I hope this information can help you guys out!

supergerrit, 2023-06-30:
Good to hear you made some progress. The flashing behavior seems very similar to the refresh command, for example, every time I want to write something to the display, I send the data bits, followed by a display refresh command. Only after this command the new information will be present on the screen.

I did some research on the Nebular 350. It seems to be a 3.5 inch 384*184 BWR display, which is quite uncommon. After some more research these displays have a SSD1685 driver [datasheet](https://v4.cecdn.yun300.cn/100001_2104235160/SSD1685.pdf). It is possible that part of the commands correspond to the IL0376F family of chips, resulting in the in the display doing something, but not what you want. I recommend you go through the commands in the datasheet and adapt you code to these, and see if the device is working.

Unfortunately I do not have access to a Nebular 350 so i cannot test anything for you. I do have access to the Nebular 154, 213 and 420, which I all managed to get working. But they all seem to use slightly different commands.

nethomas1968, 2023-06-30:
Hi, thanks for the link to the SSD1685 datasheet. I will have a read, but I notice that there is no flow chart diagram showing the steps to set things up. But, I presume the flow chart is similar the other SSDXXXX devices.
If you can send some code samples of the 154,213, and 420 for me to compare - that would be useful too.
I'm looking for the code to initialise, possibly any changes needed in the epd_display function sort of area too.

Thanks.

nethomas1968, 2023-06-30:
I suppose I should try to get the EPD_BWR_350_read_temp function working first. That only reads the temperature, and doesn't do any screen updates. That will prove that comms to the EPD are working. But, currently it's returning zero. So, that clearly isn't working yet.

nethomas1968, 2023-10-25:
[image]
I've made some progress with the Hanshow Nebular 350R-N . See picture. The epd_model is 1. So, it's being programmed as if it's the epd_bw_213 . But, still, I can sort of use it. The screen resolution needs some work of course.
I've added the code provided by supergerrit above, and changed the screen resolution to 296 X 128. I can't get it any bigger without the screen going garbled.
The next thing I want to do is eliminate the reset flashing sequence that happens when ever the screen updates. That's happening every 60 seconds currently. It's something in the EPD_Display(...) function call.
I'm hoping that I just need to tweak something to make the screen text update without the 25 seconds long flash screen sequence happening.
How can I do that?

nethomas1968, 2023-10-26:
[image]
A bit more progress with the Hanshow Nebular today.
I can only get the resolution to a maximum of Width=384 Height = 136. 384 is the full width of the screen, so that's OK. But the height of 136 doesn't fill top to bottom. There is still some unprogrammed space (see the photo).
If I go above 136 then the unit stops processing and hangs.

Also, something is triggering a full screen reset once per minute. Comes right after programming the Red bytes using Command 0x13.
I'm hoping that someone can explain how to get a seamless screen update instead of the 25 second long weird red/black/white screen flashing sequence.

nethomas1968, 2023-11-03:
Right, I've made some progress with the screen update. It's down to the full_or_partial flag.
I'm now using this code in epd_bw_213.c in EPD_BW_213_Display(..)

   if (full_or_partial) EPD_WriteData(0b11001111);
   else EPD_WriteData(0b11011111);

When full_or_partial is zero then I see that the full EPD screen reset sequence doesn't happen - and, more importantly, the screen does do an update.
There is some ghosting on the display though. And the screen update isn't exactly cleanly updating. But, it's its a big step forward.

Fin9ers, 2023-11-05:
Hello all,

New to these ESL labels and have some cool ideas of my own to try with them.
I see things are moving and active here for the Nebular series and wondered if I could be of any assistance with getting these working. 

I currently have a couple of 350f-N (Freezer versions) but they are currently sealed (will look into cracking them open) but, I can get access to a Hanshow Access point if that is of any use to anyone to progress along.

Ideally my idea will involve this access point to remotely update the ESL displays remotely.

Please don't hesitate to let me know if I can be of any assistance.

Limon93, 2023-11-05:
> D_BW_213_Display(..)

Hi @nethomas1968, can you pass me a copy of your compiled firmware?
I would like to test on my nebula ESL Tag.
Thanks!

nethomas1968, 2023-11-05:
Sure,
My latest snapshot is at http://www.nick-thomas.org.uk/Firmware-2023-11-05.tgz   . Best of luck.

YordanYanakiev, 2023-12-03:
Hello. probably an offtopic, but I got to posses quite a lot of Nebulars 1.54" and 2.6" ones.
Anyway I am trying to avoid using the Hanshow services, and I wish to post data from my own app directly to the AccessPoint.
Do You fellows by some chance came across the way the data is packed to be sent to the AceessPoint ?

MacCrutzen, 2023-12-13:
Hello, I am trying to reprogram an Nebular-420R-N but sadly didn't come far. Has someone tried one already? I also have the Nebular-213R-N and the 154R-N laying around so in the meantime I will try these 2.

void-naught, 2024-02-20:
i have a Hanshow Nebular  350R-N and i also own a flipper zero and was wondering how i should go about reading writing and data via the NFC on this display, has anyone played about with the nfc on this yet? also how would i go about recharging this display? i already tested on a samsund-duo wireless charger neither the phone or watch charging section of this charger seemed to charge this unit.

Shadowehv, 2024-04-02:
hey can som one help me i have a hanshow nebular 213R-N and would like to turn it in to a stock price notifire it has also nfc build in i already tryy to exas it but faild can you gus help me ?

Shadowehv, 2024-04-02:
here are some pictures hope you guys can help me out 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

supergerrit, 2024-04-17:
> @supergerrit By any chance, are you still able to share (preferably the .bin) what you have for the 213? I just got a few of those myself.

Here are my files. This is what i currently have for the 154 and the 430. It should be easy to adapt this for the 213 by changing the resolution settings. The 430 is a bit more difficult since the screen buffer does not fit into memory. Due to time constraints, I unfortunately had to abandon the project, hopefully some you can manage to get this working. Keep me up to date!

[154.zip](https://github.com/atc1441/ATC_TLSR_Paper/files/15012424/154.zip)
[430.zip](https://github.com/atc1441/ATC_TLSR_Paper/files/15012425/430.zip)
[image]

MrDeon, 2024-11-09:
I've managed to flash a couple of Nebular 346 units successfully, but I have a few other models too that I can't find firmware files for. Is there support for any of these Nebular devices?

- 266Y-N
- 266CY-N
- 420Y-N
- 583Y-N

valtoree, 2024-11-28:
Hi @nethomas1968 , you've made amazing progress with 350r-n, I too own one, curious if you've gotten to a usable state. Was hoping to have a daily reminder made out of it. Tx

nethomas1968, 2024-11-29:
Sadly, I've somewhat given up on this project. I couldn't get the Hanshow
screen to work reliably, and I don't know enough about the firmware, or how
these devices work to make it work properly.
I was hoping to have a small cheap display that I can use on top of a
Raspberry Pi in an embedded MP3 player project.
But, I couldn't get this EPD to be fit for purpose.
And, I've moved on to other things.

I do keep an eye out on some other forums and github projects - but unless
someone uploads a working firmware for the Nebular 350R-N then I won't be
using up any more time on this.

Christmas is coming, so I might just buy a display that is known to be
working with a modern Raspberry Pi.

Regards

Nick

tobiasdroessler, 2024-12-04:
@supergerrit   I take your 154.zip and applied (overwrite) the src files from that archive into the current state ot this repo ... compiles and build runs successfully  👍 ... but when i flash it the device will not work 👎 ... any hints/ideas?   I mean the result in your picture looks very nice for the small 154 device.

Casperdroid5, 2025-03-11:
Hello everybody.
Has any progress been made on the Nebular 2.13R-N models?
I have quite a few, but I can't get them to refresh/change the screen. The LED colour does change from time to time, depending on the software I flash it seems.

kero3009destiny, 2025-03-15:
Just got my hands on 2 Polaris Pro labels and they seem to be using the same package. Hoping to dig deeper, same methodology might as well work for this device?

[image]

RadioactivePotato, 2025-04-27:
Does anyone know the baud rate that it uses to communicate over UART, most of them just returns back gibberish.

nethomas1968, 2025-04-27:
Baud rate is 115200. There is a uart.c file that has that as a comment.
No Parity, 1 stop bit.

shugga, 2025-09-21:
Hi@all!
I, as a not-so-technical person, asked myself if is also possible to write the firmware with a Raspberry Pi?

AdamGorniak, 2026-05-03:
Hi, i have Nebular Pro 437Q-N, with the HS9118 on board. I try a few binares, but with no results. I flash with the Uart Ftdi programmer. Pinout is:  3,3V >VDD, Gnd>gnd Txd > SWS and Rst > Rst. This od correct?  Soneone flash this model with success?