atc1441/ATC_TLSR_Paper #19: Temperature format is incorect! must be int8_t and not uint8_t how it is in these sources

raduvas, 2023-01-07:
Hi,
Thank you for this fantastic SDK!

I make a paragliding variometer for me and I noticed a mistake for the temperature in the source code.

Basically the temperature must be  int8_t everywhere instead of uint8_t to be able to get sub zero temperatures otherwise for -1'C you will get 255'C (for me I corrected it in "epd_bwr_154.c" and "epd.c")

**IMPORTANT** What do you use to read/write BLE on the phone? I cannot get the app name from your videos.

[temperatore register in datasheet.pdf](https://github.com/atc1441/ATC_TLSR_Paper/files/10367069/temperatore.register.in.datasheet.pdf)

Sterillium, 2023-01-08:
Just an amateur here... Thank you so much for your post! I've been struggling with this exact issue. Your project sounds really cool! Would you consider sharing pics and/or the code?

raduvas, 2023-01-08:
I modified for you the temperature from unsigned to signed and I attached the "src" directory here.

I did modification to these files:
epd.c
epd_bw_213.c
epd_bw_213_ice.c
epd_bwr_154.c
epd_bwr_213.c
epd_bwr_350.c
epd_bwy_350.c

Make a backup of your work and after that replace the src directory of what you have with the one extracted from my attachment from here.
[src.zip](https://github.com/atc1441/ATC_TLSR_Paper/files/10369252/src.zip)

raduvas, 2023-01-08:
To be clear:
1) Clone and download this github (ATC_TLSR_Paper)
2) Extract on your hdd
3) Download my attached src archive and extract and replace the src directory of ATC_TLSR_Paper with my src directory
4) Install Python to be able to compile the example
5) execute in firmware directory "makeit clean"
6) execute in firmware directory "makeit"

You will get "ATC_Paper.bin"

raduvas, 2023-01-08:
I attached a picture of my work in progress:
- first line is the up/down paraglider speed, must be easy visibile and values are between -20.0 m/s and +20.0 m/s
- second line is barometric height (in meters) from the sea level
- bottom line has temperature and battery voltage.
In the above src the first and second line are static now. In the future they will be get over UART RX TX from my variometer made with ATmega and Bosch barometric sensor.
[image]

Sterillium, 2023-01-08:
Dude! You rock! I finally understand it. Thanks for sharing and the good explanation. That's gonna be a cool little variometer!

In your source files I did have to change uint to int (for EPD_BW****_read_temp and EPD_BW*******_Display) in all the epd_bw****.h files to get it to compile with makeit.exe. Now it works flawlessly for me. Thanks!

raduvas, 2023-01-09:
Yes, I changed all uint to int in the sources and I did not test compile the sources before attaching them here. You are really good too if you solved those errors by yourself :)