atc1441/ATC_TLSR_Paper #20: Firmware not working on Stellar-XM3YN@ E31A

Drewnja, 2023-01-18:
Hi, included firmware doesn't display anything and doesn't advertise BLE. My pricetag has BW/Yellow screen. I have not seen those in the discussion here. 
Thanks
[image]
[image]
[image]

MrHadess, 2024-12-19:
My model "XM3N@---E31A" works fine in this project, maybe you can give it a try

Project Link: [stellar-L3N-etag](https://github.com/reece15/stellar-L3N-etag)

zhaoqi90, 2025-01-10:
How does the screen display，Same model but incomplete display

MrHadess, 2025-01-10:
> How does the screen display，Same model but incomplete display

Are you using this project [stellar-L3N-etag](https://github.com/reece15/stellar-L3N-etag) and is the model 'XM3N@---E31A'? If yes, can you show your device state?

zhaoqi90, 2025-01-11:
The screen of 2.66, with a resolution of 296/152. Brushing in the splash screen does not display the EPD BWR213, but it does display but the resolution is incomplete. 
[image]

zhaoqi90, 2025-01-11:
The model is consistent, but the display is not perfect. Do you have firmware or guidance to modify the resolution

MrHadess, 2025-01-11:
> The screen of 2.66, with a resolution of 296/152. Brushing in the splash screen does not display the EPD BWR213, but it does display but the resolution is incomplete. ![2c27d69d24aaad637930361be14bdf9](https://private-user-images.githubusercontent.com/127290223/402195306-42a581ab-01b2-48a1-ab78-c50e179cee7c.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY1NjgzNzIsIm5iZiI6MTczNjU2ODA3MiwicGF0aCI6Ii8xMjcyOTAyMjMvNDAyMTk1MzA2LTQyYTU4MWFiLTAxYjItNDhhMS1hYjc4LWM1MGUxNzljZWU3Yy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDExMVQwNDAxMTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zNWQ1MGUzMDI0MTEzZWJjOGVkZDM1MzMzMmFlNTcxZGIyMDE1Mzc1YTYxNDViOGY0OTk4YjgyMmRmZTBlY2VjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.EFFsmBy0PaOmo2e0KkdF31CIzv0xYZkm8PVW4Zeq_uY)

Yes, I also think that this firmware may not adapt to this resolution. Maybe you can change the source code to fix it.
Oh...I'm not developing code on this MCU board, I'm trying anothe dev borad to control this screen

zhaoqi90, 2025-01-11:
Okay, Okay,thank you