atc1441/ATC_TLSR_Paper #22: M3@ Pinout changed

crazy888, 2023-01-18:
[image]

I think it´s the some chip but what is my Tx line for flashing?
Someone knows?

atc1441, 2023-01-18:
Hello.
That version is not supported, it has a different SOC
Only TLSR version is supported

crazy888, 2023-01-18:
Okay i messed it up................. Thanks for quick reply

8bignic8, 2023-01-22:
I have the same display, does any one know what kind of Controller is on the Board? Or does anyone know how to falsh that badboy :D?

8bignic8, 2023-01-22:
i found this: https://github.com/CursedHardware/hanshow
and my serial number on the Board leeds me to this: https://github.com/CursedHardware/hanshow/blob/master/HS_EL5102.md

atc1441, 2023-01-22:
The chip on it is an MSP430
It is an 8051 core SOC and does only support propriatary RF protocolls, no Bluetooth, just as info

8bignic8, 2023-01-22:
But  could still hook it up to esp8266 and write to it?

atc1441, 2023-01-22:
Yes that is possible

crazy888, 2023-01-22:
> Yes that is possible

The question is how :) ?! 
Soldering the pins is impossible. The SOC is so small and i dont know even where and how to connect the ESP.

8bignic8, 2023-01-22:
theoretically you got vcc 3,3v gnd Ground, DA data?, ck clock? , cs chip select, nrst reset?

crazy888, 2023-01-22:
[image]

[image]

crazy888, 2023-01-22:
> theoretically you got vcc 3,3v gnd Ground, DA data?, ck clock? , cs chip select, nrst reset?

But it´s controlled by the chip on board already? So you have to desolder the existing SOC?

8bignic8, 2023-01-22:
no you can use the i2c or uart bus on the other side. from your first picture at the top of the issue

8bignic8, 2023-01-22:
theoretically you got 
- vcc 3,3v 
- gnd Ground, 
- DA data?, -
- ck clock? , 
- cs chip select, 
- nrst reset?
- https://www.youtube.com/watch?v=r-N53fpaJWo

atc1441, 2023-01-22:
You need to create a custom firmware that you flash on the MSP430 or you need to desolder it and solder to the screen pins.

But both options are not part of this repo i am afraid.

Here is an example of soldering to the display pins:
https://youtu.be/RUoMztFxUpE

8bignic8, 2023-01-23:
Have a look @ https://cursedhardware.github.io/epd-driver-ic/SSD1675B.pdf
in section 6.1.2
>> The 4-wire SPI consists of serial clock SCL, serial data SDA, D/C# and CS#. The control pins status in
4-wire SPI in writing command/data is shown in Table 6-2 and the write procedure 4-wire SPI is shown
in Table 6-2

8bignic8, 2023-01-23:
> 

if found the Datasheet for the antenna: 
https://datasheet.lcsc.com/szlcsc/2001071135_AMICCOM-Elec-A71X06AQFI-Q_C479510.pdf

maybe we could build upon that :)