atc1441/ATC_TLSR_Paper #25: Not using whole display, forcing EDP not working

sietseharteveld, 2023-02-22:
Hi! I've been tinkering with this display for some time now. The firmware works and I can flash firmware using Bluetooth. I have one problem: the device is not using a big part of the screen and I am not sure whether it is using the right EPD model. Forcing different EDP models via the bluetooth flasher does not do anything. The device is: Stellar-MN@ E31HA, so the 200x122 version. Here is a picture of the display with the problem visible.
[image]

sengelsman, 2023-02-23:
I have some of those Stellar-MN@ E31HA and made it work to use the full screen. You can find it here [stellarmn@](https://github.com/sengelsman/ATC_TLSR_Paper/tree/stellarmn%40)

mihaiolimpiu, 2023-04-15:
@simon-src: It worked great, but image upload was cutoff, I also added 250/128 as width height as in this link to your firmware:
https://github.com/atc1441/ATC_TLSR_Paper/issues/16
(but my BT is a little spotty I noticed, I need to retest when I'll have some time to be 100% sure it wasn't anything else)

alex12071986, 2024-01-03:
> I'm attaching the pre-built binary with Stellar-MN@ E31HA support. Built from cleanly checked out [stellarmn@](https://github.com/sengelsman/ATC_TLSR_Paper/tree/stellarmn%40) [ATC_Paper.bin.gz](https://github.com/atc1441/ATC_TLSR_Paper/files/10935492/ATC_Paper.bin.gz)

Hello,
Do you have the source for compiling that ATC_Paper.bin.gz? Because that firmware works, but if i want to compile the one from here: https://github.com/simon-src/ATC_TLSR_Paper/tree/stellarmn%40 is not on the full screen.
Thank you.