atc1441/ATC_TLSR_Paper #31: Hanshow Nebular - Pinout.

nethomas1968, 2023-03-29:
I think the Hanshow Nebular has different pinouts for the GPIO lines. I notice that the SPI comms are not working for the LUT tests.
So, in main.h I see:
#define EPD_RESET 	GPIO_PD4
#define EPD_DC 		GPIO_PD7
#define EPD_BUSY 	GPIO_PA1
#define EPD_CS		GPIO_PB4
#define EPD_CLK		GPIO_PB5
#define EPD_MOSI	        GPIO_PB6
#define EPD_ENABLE	GPIO_PC5

But, some or all of those are wrong. Is it possible to obtain the correct pinouts for this device ?

nethomas1968, 2023-03-31:
Well, I managed to flush out the GPIO pins on a dismantled Hanshow Nebular 350R-N , and the GPIOs are set up as shown above.  Note that EPD_ENABLE and EPD_BUSY are inputs in to the Hanshow chip.

j0nnymofo, 2023-11-06:
Hi @nethomas1968, any possibility you could post a picture of the pinout? Is it the small pins on the front or the larger pads on the rear of the board?

nethomas1968, 2023-11-07:
[image]
Hi, yes, in this thread I'm talking about the small contact points on the back of the PCB. These are the connections to the ribbon cable that connects to the actual display. Photo uploaded.
I did flush out those pins, and yes it's all connected up as described in main.h .