atc1441/ATC_TLSR_Paper #35: Hanshow Nebular - EPD_BUSY is constantly high.

nethomas1968, 2023-04-12:
Hi,
I'm debugging the Hanshow Nebular 350R-N. I know that this model isn't currently supported with the current firmware - but the firmware does load and run, it just doesn't do anything with the display.

I'm looking at how far it's getting. I notice that EPD_BUSY pin is set to input, and it's constantly measuring a high (logic 1). I presume that an initial attempt to do something with the display is making the EPD busy, and then we are waiting for that operation to complete before doing anything else with the display.
But, with EPD_BUSY being constant high no further updates are attempted.

What can be the reason?

nethomas1968, 2023-04-12:
Upon further inspection, EPD_BUSY changes state in this piece of code:

    // system power
    EPD_POWER_ON();

   // NOTE: EPD_BUSY is low here.

    WaitMs(10);
    // Reset the EPD driver IC
    gpio_write(EPD_RESET, 0);
    WaitMs(10);
    gpio_write(EPD_RESET, 1);
    WaitMs(10);

   // NOTE EPD_BUSY is high - and never changes to low again !

hitesh-singh, 2023-05-04:
@nethomas1968 Did you manage to get any further on this ?

nethomas1968, 2023-05-04:
OK, I've got another Hanshow Nebular opened up now, and it behaves the same the other one. EPD_BUSY never goes low after a call: gpio_write(EPD_RESET, 1); 
I can only assume that:
1) The Hanshow Nebular 350R-N uses a different sort of E-Paper display device that requires a different comms protocol.
2) The MCU chip in the Hanshow Nebular has a driver chip that isn't SSD1675/SSD1680 compatible. 
3) Something else.

Any help appreciated.

nethomas1968, 2023-08-22:
Hi valtoree.
It's not easy to open it at all. I used a hacksaw on the back. I completely mashed up the first attempt, so I ended up dismantling it entirely. It was a horrible mess, and I also cut my fingers. So, be careful.
For the second attempt I tried to just saw the corner. So, the LED on the front is in the top right hand corner, I used the hacksaw on the back at that same corner. That's where the little circuit board is located. It's probably better if you have a few of these devices, because I've completely destroyed 5 of them while trying to get access to the back progamming pins.
My best attempt involved using a drill - and very carefully push down to drill through the plastic. Stop before you hit any electronics - but that's the tricky part.
I knew where to drill because I had already completely dismantled several other of these devices before trying the drill method.

The short answer is - there is no easy way to gain access to the progamming pins.

Best of luck!

valtoree, 2023-08-22:
@nethomas1968 tx 4 reply. I will wait for a NFC hack 🤷🤞

philenotfound, 2023-10-31:
@nethomas1968 I ended up at the same point (BUSY stays high after RESET), then discovered you have encountered the same issue. Have you made any progress or found any data sheet so far?

nethomas1968, 2023-11-01:
Hi @philenotfound  . I have made quite a lot of progress with the 350R-N. But, that's only since I made sure to set the epd_model to 1. See the EPD_detect_model function in epd.c . So, the code is using functions in the epd_bw_213.c file.
The advice to use the SSD1685 data sheet appears to be wrong - not much in there works for me.
Whatever chip is used in the BW213 appears to be the same as the 350R-N Nebular EPD.
I've uploaded my current firmware to http://www.nick-thomas.org.uk/nebular-firmware.tgz . It's a bit hacky in there, but hopefully you will see what I've done to get this far.

philenotfound, 2023-11-01:
Thank you, I'll have a look!

rubeycubey, 2023-12-22:
Re opening the Nebular 350R-N: I used a fine spudger to pry the screen off the front.  There is a layer of clear plastic and then the e-ink screen.  I think I got the spudger behind both the clear plastic and the screen which resulted in me cracking the e-ink screen.  Next time, I would use a heat gun and try and remove just the clear plastic screen with a suction cup.  The screen and circuit board should just drop out the front then, leaving the battery behind.  Taking the front off doesn't require damaging the case.

g0emf, 2024-01-06:
> Any progress on the Nebular 350R-N:
Any chance of a picture of the board & did the suction cup + heat gun work?

rubeycubey, 2024-01-06:
I haven't opened another yet - waiting for progress on replacement firmware.  There are pics uploaded elsewhere, but attached are pics of the one I opened.
[image]
[image]
[image]
[image]
[image]
[image]

g0emf, 2024-01-07:
Hi thanks for all the pictures, it took me some time to find information for Nebular 350R 
I have checked  http://www.nick-thomas.org.uk/nebular-firmware.tgz but the link does not work, ( http://www.nick-thomas.org.uk ) is working but nothing found on the Nebular 350R.

Whats the the source to monitor for any updates?

Thanks again

nethomas1968, 2024-01-07:
http://www.nick-thomas.org.uk/nebular-firmware.tgz is there, and I've just downloaded it. What tool were you using to download? I simply typed that URL in to the browser bar and it started downloading !

g0emf, 2024-01-07:
Thanks Nick,

My browser was blocking it LOL

Keep in touch

Regards

Dave

nethomas1968, 2024-01-07:
I've uploaded a newer version of the Nebular firmware. The new one still has lots of problems with the actual screen display, but it does have a serial command line functionality. So, you can user serial port to communicate with the nebular (if you've got the RXD and TXD pins connected up). It makes it a bit more interesting to play with.
The idea is that I will be able to enter some debugging commands - when I get around to adding them.

The new firmware is http://www.nick-thomas.org.uk/nebular-firmware2.tgz 
Baud rate is 115200.

Best of luck.

g0emf, 2024-01-07:
Understood Nick.
Thanks for your help and efforts.
Will see if I can get inside to get the interface up and running then have a play ;)

atc1441, 2024-01-07:
A short flashing video for that type is now online here: 
https://youtu.be/WRfZJ4xyYwc

This shows just the flashing without the Firmware being released right now but will still help.

nethomas1968, 2024-01-07:
Oooh, yes, that looks interesting. The project at https://github.com/jjwbruijn/OpenEPaperLink/tree/master/ARM_Tag_FW/OpenEPaperLink_TLSR looks very interesting. But, it needs the tc32_linux directory adding before the makefile works on Linux builds. But, I guess that's in progress.

g0emf, 2024-01-07:
Thanks, Nick I will have a look at that when I get a min.... too many projects on the go!

g0emf, 2024-01-07:
Hi Nick, I just spotted this (just been uploaded)

https://youtu.be/WRfZJ4xyYwc?si=G2AzVNc8a7_iFFFg

nethomas1968, 2024-01-10:
What I don't understand here is that the Hanshow Nebular 350R-N is a 3.5 inch display with 384x184 resolution. It's a BWR display device. The source code in epd_bwr_350.c shows that it's controlled by a SSD1685 (or similar) EPD controller. Now, that EPD controller data sheet does show that it's capable of 384x184. But, that code (when running) doesn't do anything to my device.
But, if I use code that is intended for the UC8151C EPD controller then I do see something on the display. But, the resolution is wrong.
So, I'm inclined to believe that there is a UC8151C (or similar) chip in the Hanshow Nebular 350R-N . But, the code isn't setting the resolution properly.
Is there a newer version of UC8151C (with a data sheet) intended for this bigger display ? I see that the UC8253C is similar, but the data sheet doesn't show 384x184 as an option.
How can I deduce what EPD controller chip is actually in use ?

Mexmaster00, 2024-01-21:
> What I don't understand here is that the Hanshow Nebular 350R-N is a 3.5 inch display with 384x184 resolution. It's a BWR display device. The source code in epd_bwr_350.c shows that it's controlled [...]

it seems that not all BWY are supported, not all of them use the same EPD Driver. but i didnt find anything about that.

g0emf, 2024-01-22:
I would like to comment with some useful information on this but I am not up to speed with epd's
It could well be different hardware used with the same model number?
Any numbers on the chips?

Mexmaster00, 2024-01-22:
> I would like to comment with some useful information on this but I am not up to speed with epd's
> 
> It could well be different hardware used with the same model number?
> 
> Any numbers on the chips?
> 
> 
Maybe that is helpful 

[image]

mcauser, 2024-04-08:
Anyone know what that white square 4-leg component is in the top right?

mcauser, 2024-04-10:
🤦 There's a hole in the PCB and it's reverse side mounted. _My bad._

dontrajik, 2024-06-11:
I get TIMEOUT WAITING FOR EPD_BUSY at Point 2
This is sort of bricking my tags.

stematon, 2024-06-14:
Hi @nethomas1968 , I can confirm that your modified FW runs the 350R-N. But my version differs from yours:
[image]
I had a LiPo battery (I removed it because of mechanical stress during disassemble)
Anyway the device is alive:
[image]

But I don't see debug messages. Can you give your RS232 settings? 115k2, 8N1?

nethomas1968, 2024-06-15:
Ha ha, that's cool.
But, yes, the 350R-N that you have does look different. The programming
points are in a different place!

Fun to see my name on your EPD :)

Regards

Nick