atc1441/ATC_TLSR_Paper #37: BLE connection with Wemos or Android?

condor13, 2023-05-19:
How could I achieve a connection with an Wemos ?! would like to convert this to a strange looking "smartwatch" (just receive text notifications on it)

nethomas1968, 2023-05-22:
I don't think the Hanshow devices are WiFi, but they have Bluetooth capabilities. And, I don't think the Wemos devices have bluetooth. I suspect that means that will have to connect the two devices together using old fashioned wires. The Hanshow devices can be communicated with using a serial interface. The Wemos has a number of GPIO pins that, I presume, can be configured to work in a serial manner. That would be my starting position I guess. I don't have any code examples on how to use GPIO pins on the Wemos as a serial port though.

condor13, 2023-05-23:
Yes.. I know Hanshow devices dont have wifi, but you make a good point.. I could wire a serial connection. I know how to wire it so serial and how to write code to send serial from arduino.. but were should i look in this firmware in order to know how should I talk to it via serial.