atc1441/ATC_TLSR_Paper #40: Stellar Pro-213R-N - firmware flashing failed

plouflechien, 2023-07-31:
Hello, I have some not well known "Stellar Pro-213R-N "

[image]
[image]
[image]
TLSR8359 seems to be elligible to be hacked, but the screen is 250x122 and it has 3 colors (BW+red).
Connecting to serial, seems to work (blue led is flashing while sending commands to it), but I can't say if it's working : any hint to read any output/debug from this ESL ?

Beer101010, 2023-08-19:
Hello !
The chip is compatible with the custom firmware, but the screen driving part isn't working as it should.
I've got a bunch of these, and have the same problem. 
The flashing works, for debug, you can use the TXD pin on the ESL to the RXD on the USB flasher. 
Tried changing model with e00X over BLE does work, but the screen doesn't update with any model number.
I've tried to debug it a bit, but could not guess what was wrong with the SSD1680 / 1675 addresses. Or if it's the right place to look. Sadly i'm not good enough at coding to figure it out.

Some other models won't work either for the same reason, like the nebular. ATC1441 said he's too busy to work on it at the time, hopefully someone will make it work one day.

I Have a bunch of smaller Stellar Pro-154-NP (same series as the 213-N) that do work without issues !

And Thanks to ATC1441, and all the people who worked on this, i have a blast playing around with these ESL, i'm using them with an RPI to display a temperature probe on my beer tanks at my brewery ! 

Cheers !

thxomas, 2023-12-27:
Hello !
I got my hands of a bunch of Stellar Pros too. I managed to flash and use the BLE features. 
However the screen does not update as mentioned in OP.
So I've been digging info and tinkering. Thanks to the work of all, here's what I managed to get :
- I first checked if the GPIO pins were the same : it looks like EPD_ENABLE is wired to GPIO_PB7 instead of GPIO_PC5. (I'm not sure but PC5 looks not wired and the PB7 pin looks to be routed to the capacitor part of the board that leads to VDDIO pins)
[image]

- Based on the references on the back of the screen, I also found the screen controller has high chances to be JD79656 (SSD1680 compatible)
- While debugging the firmware with some added serial logs : 
   - The main loop runs well
   - As mentioned in other issues the BUSY pin is high most of the time 
   - I tried all the models supported by the firmware : the screen never updates anything.
- I started comparing datasheets and commands sequences but to a dead end at my current skill level.

So, before sharing this, I tried another approach : I extracted the original firmware, and try to reverse it, in the hope of getting some low level info on how the screen is driven. This task has also hurdles, but so far I managed to load the code in ghidra, disassemble it using [https://github.com/trust1995/Ghidra_TELink_TC32/tree/master](url) and start understanding the structure of the code (I think the same SDK is being used, I found similarities in the objects layout). However, there's still a lot to be done.

I just wanted to share this, just in case, and in  the hope that someone will revive the subject with me, or get some advices on the best way to figure this out.

Cheers !

ssdollxx, 2024-02-15:
Hello,
https://github.com/Reletiv/OpenEPaperLink_TLSR
in folder [Stellar_2.13_BWR_PRO] are working files for Stellar Pro-213R-N

belkone, 2024-08-23:
> Hello, https://github.com/Reletiv/OpenEPaperLink_TLSR in folder [Stellar_2.13_BWR_PRO] are working files for Stellar Pro-213R-N

Hi!

Did you just flash the file from https://github.com/Reletiv/OpenEPaperLink_TLSR/blob/main/Stellar_2.13_BWR_PRO/STELLAR_PRO_2.13_BWR.bin and it worked with Stellar Pro-213R-N?

After flashing, I can't see the BLE device, the screen refreshes showing white letters on a black background (which is the opposite of everywhere I've seen) with the content AP Found...

belkone, 2024-08-23:
> Hi!
> 
> I did not use the binary directly. Instead I merged parts of the code (mainly the screen driver) with my own source.
> 
> Attached is my working version. It supports the web tool for OTA and sending images including in 3-color. The BLE device name should begin with THX_. This firmware IS NOT the OpenEPaper version compatible with AP, but the original one from atc1441 working only on Stellar Pro 213R,
> 
> I provide this with no further support as I stopped research on these devices.
> 
> [ATC_Paper.zip](https://github.com/user-attachments/files/16728522/ATC_Paper.zip)

Hi, 
sure! thanks a lot for the answer and the binary :)

belkone, 2024-08-23:
@thxomas I have two more questions if I may:
1. which exact firmware for Stellar Pro 213R from atc1441 do you mean, can you send a link to the repository, please?
2. should your firmware looks like the one in the picture?
[image]

thxomas, 2024-08-24:
1. The firmware is from the repository that we are discussing in : https://github.com/atc1441/ATC_TLSR_Paper
2. On mine this screen has black text on white background. I don't know why it looks inverted on yours. However it is unfinished, I focused on displaying 3-color images : 
 
[image]

If you're interested I may share my codebase

belkone, 2024-08-24:
Ok, that's what I thought, but I wasn't sure, because you mentioned old software for Stellar Pro 213R and it's not visible in the description of this repository and I thought that there was another repository :)

Sure! you can share it, maybe I will fix it for my tags, if not me then surely someone will find it useful!:)

Fred2204, 2024-11-22:
I managed to flash a custom firmware to the epaper using this flashing tool: TLSR8359 USB-COM Flash Writer v0.2 (TX-SWS only!)

Now the epaper is showing a clock which is updating every minute. However, it seems I cannot flash anything new now. I tried to unlock and erase the flash but it keeps updating the clock even during the flashing process and it seems I am unable to flash any new code. Any ideas what I could try?

Fred2204, 2024-11-27:
@thxomas
Hi, I was hoping you could help me a moment.
I flashed the binary in ATC_Paper.zip in your comment of 23rd of August onto my Stellar Pro Epaper and it is working fine.
I was hoping that the binary in "https://github.com/reece15/stellar-L3N-etag" would also work on my epaper but it doesn´t.
Would it be possible to for you to share your codebase to create the ATC_Paper.bin from you comment of 23rd of August? You offered this to @belkone  in your previous comment but it seems at the end you didn´t do it.

thxomas, 2024-11-27:
Hi !
You're right, I did not follow up my sharing proposal. Better late than never, here it is ! 

[stellar3_etag_bwr_thxomas_src.zip](https://github.com/user-attachments/files/17928872/stellar3_etag_bwr_thxomas_src.zip)

(the zip includes only the src folder, to be merged with your local repo of the @reece15 project).

FYI :
- All the webtools of @reece15 work with this version (OTA to send images (incl. Black/red), bluetooth OTA upgrade firmware, etc...)
- The image display mode is working, ("Set to picture mode" in the webtool)
- The "standalone" clock screen may be broken because I was in the process of making it black&red, but did not finish (lack of memory on the device to handle buffers for each color plane). You should be able to fix it. 

I'm happy that this work still generates interest and I may end up setting up a correctly forked repo on github.

dbcooperthree, 2024-12-21:
Hello!

Hello Guys! I have this label from the pictures. 
[image]

I printed an adapter so I wouldn’t have to solder the wires, and I have the two USB TTL adapters. 

[image]

[image]

[image]

Somehow, I flashed the firmware shown in the picture, and now I can’t flash anything else. 

[image]

In the past, the blue LED would light up when I flashed something. Now it doesn’t. I’ve been reading a lot, but I’m not sure which is the correct firmware for the type of label I have. Also, I don’t know where to connect the wire coming from the label (Rst) to the adapter. Can you help me? 

Thanks a lot!

znawcaphotona, 2024-12-22:
Hi,
I have this ESL but after uploading the firmware nothing happens - no Bluetooth and nothing on the display. I tried the default 460800, 3s and as suggested in the comment on the YouTube video 921600, 0ms. The blue LED blinks every time but still nothing. I have tested two of them and both do not work. I think the reason may be that I am using a nodemcu (esp8266) with en connected to gnd so I can use it as UART but it worked with other devices. Is there anything else i could try to flash this tag? I tried everything suggested here.
[image]

Fred2204, 2024-12-23:
For me the flashing works with Baud 115200. I set the Atime to 8 secs to give me more time. After hitting "unlock flash" I quickly connect Rst to ground for a moment. Then I hit "flash firmware" and quickly connect the Rst to ground again for a moment. This way it is working for me. It is important to use the src folder from thxomas to replace the src folder in the atc1441 code base. Otherwise the Pro version of the epaper was not working for me.

dbcooperthree, 2024-12-23:
If I replace the src folder from thxomas in atc1441’s code, make.exe stops right after this line. I managed to replace it with the files from here: https://github.com/reece15/stellar-L3N-etag. However, I still couldn’t get it to work. I think there’s an issue with the OpenEpaperLink firmware, and I don’t know how to remove it. The blue LED no longer blinks when the red TXD LED on the adapter flashes. On one of the adapters, the red TXD LED doesn’t light up at all if it’s connected to the ESL.

thxomas, 2024-12-23:
Hi,
I had similar issues with my first tries.
Luckily I had a Raspberry Pi that I could use with this low-level TLSR Flasher :

[https://github.com/pvvx/TlsrComSwireWriter](url)

The cabling is slighlty different and it makes use of the RTS pin to reset the chip and flash before it boots the firmware (RTS is not available on all USB-to-UART adapters). I remember having to enable the RTS/CTS feature on the Pi too.

it requires some hardware you may not have at your disposal but I remember being stuck like you with an ESP board (worked only once).

Hope this helps
T.

dbcooperthree, 2024-12-23:
I actually have a Raspberry Pi 4 Model B. Unfortunately, I can’t access the link you attached. Thanks a lot!

znawcaphotona, 2024-12-23:
I also have Raspberry Pi 4 Model B, but this tool doesnt work as expected:
[image]
It shows nothing more than just "error". I also did `chmod 777 /dev/serial0` but still nothing. I know its not right repo for support on this tool, but did it work for you? Also typing `echo test_message_hello_esl > /dev/serial0` makes blue LED flash so everything is connected right

thxomas, 2024-12-23:
> I actually have a Raspberry Pi 4 Model B. Unfortunately, I can’t access the link you attached. Thanks a lot!

Edit : corrected the link in my previous post

dbcooperthree, 2024-12-23:
[image]

I finally managed to do it. I succeeded with one of the adapters that also had the RST pin. I was able to get ATC_Paper.bin as mentioned in the recommendations here, but it still doesn’t work. Do I need to do anything specific after flashing the firmware?

belkone, 2024-12-24:
Hi,

if you want to use with Open Epaperlink then below fork works well with Stellar Pro 213R-N:

https://github.com/Reletiv/OpenEPaperLink_TLSR/tree/main/Stellar_2.13_BWR_PRO

dbcooperthree, 2024-12-27:
Hello! I finally managed to do it. I think the problem was with the browser I was using. Chrome is the right one. However, I still have issues with the colors, but I think it’s good enough for me. I believe the issues are already known. However, I’m attaching some pictures. Thank you very much for your help.

[image]

[image]

[image]

Fred2204, 2024-12-30:
Hi there,
does anyone know how to speed up the refresh time of a full update or how to carry out partial updates on the Stellar Pro-213R-N? The codes I found here for similar displays which are including the partial update mode, seem not to be working on this display...

lleomr, 2025-02-08:
Hello everyone !
I managed to flash the firmware from https://github.com/Reletiv/OpenEPaperLink_TLSR on my Stellar Pro 213R-N but the bluetooth seems to be not working.
I tried to display a clock or an image using https://javabin.cn/stellar-L3N-etag/web_tools/ but I can't find my ESL on the list (and can't manage to find it with the MAC adress, any tips on this would be appreciated !)
Thank you by advance for your time and your help !

dbcooperthree, 2025-02-08:
OpenEpaperLink doesn’t work with the tool you want to use. Read the comments above. thxomas explained it very well, and it worked for me.

lleomr, 2025-02-08:
Thank you for your fast answer !
So if I understand well I need to merge the firmware from https://github.com/reece15/stellar-L3N-etag with the code provided by thxomas ?

jofr666, 2025-03-19:
Hi,
I have a Stellar Pro-154R-NP.
Has anybody tried one of these?
I was able to flash it but I can't connect to it with ATC_BLE_OEPL_Image_Upload.html.

Beer101010, 2025-03-19:
I have a few working
They should work out of the box, but sometimes it takes multiple tries to
flash it properly.

When, flashed successfully the green led should blink every few seconds /
10 seconds.
Also, you have to push the value e003 IIRC when BLE finaly works, so it
uses the right resolution for the screen. By default the firwmare is using
the 213 model (e001).

jofr666, 2025-03-19:
Thank you for the fast response.
I'm getting a clock on the display ( size is not correct) but when I want to connect with BT it never connects.
I have not seen a green led but the blue one blinks about every 30 secs and the clock is counting.
 I flashed the fill from Thxomas on it.

dene-, 2025-08-06:
> Hello, has anyone managed to get this model working?
> 
> [image]

I have almost the same model and even though I can flash it and it's detected on the USB Serial Port and all, the blue led never turns on and it never resets either. Weird because the flashing process happens...