atc1441/ATC_TLSR_Paper #41: Stellar-MFN@ E31A not working properly / Display won't change.

ValiPR, 2023-08-07:
Heya all!

I'm kinda new to all this so not sure where it went wrong or if my device just isn't compatible.

I flashed the ATC_Paper.bin to the Stellar-MFN@ E31A but the screen isn't doing anything and I am not able to chance anything, I am able to connect to it over bluetooth (ESL_FC09DB) But that's about it
[image]

If anyone has some ideas or could offer some help I would be really grateful!

obones, 2023-08-22:
I'm using a MFN@ E31HA here and the result is about the same.
The display did get change a little bit, some white lines appeared in the middle of the existing price as displayed when it was in store, and there is a black rectangle flashing at the border of the display.

How should one go about sending an image to this display?

ValiPR, 2023-08-22:
> I'm using a MFN@ E31HA here and the result is about the same. The display did get change a little bit, some white lines appeared in the middle of the existing price as displayed when it was in [...]

I actually managed to order a MN@ E31HA hoping it would be different and the result is I can flash it but the flash is "luck based", it shows the clock sometimes, but it's off center, using other users flashes nets the same result sadly :(

lvjiaming0418, 2023-09-22:
My m3n@ even doesn't have the bluetooth.
I cannot connect with it except the USB-COM Flash.

TheoHarbers, 2024-06-23:
I have the same thing Stellar-M3N@ E31HA sn 301119. Flashing was without problem. I can talk to it over BLE but the display does not do very much. Sometimes after i sent some commands  and after 30 seconds or the display seems to produce a dark edge (couple of pixels wide) all around the outside stays visible and seems flashing a couple of seconds. I have a black-white-red version. I have tried the commands to change the display type manually, 0xE0020 or xE003 force set an EPD model, if it was not detect automatically correct. But no luck with that. any ideas are welcome.