atc1441/ATC_TLSR_Paper #42: Hanshow Nebular Reset Flashing Sequence.

nethomas1968, 2023-08-10:
Hi,
Sometimes my Hanshow Nebular 350R-N will go in to a special flashing sequence which last about 20 seconds.
The whole screen will flash from black to white. The flashing slows down, and changes to red then white.

I've seen it do this when I've used some incorrect commands in the firmware code that I'm playing with.
But, I can't reliably make it do that through software.

Is there a specific "doTheResetFlashingThingNow()" function on these devices?

nethomas1968, 2023-11-02:
After some more research, it looks like the command that is triggering the screen reset sequence is:
EPD_WriteCmd(0x12);

This is on the Hanshow Nebular 350R-N.

I'm not at all sure which EPD chip is in there. Some references show that it's an SSD1685 based chip, but non of those commands in that data sheet work.
The commands from the UC8151C and IL0376Fb93d do (sort of work) - in that I can get some text to appear on the screen.
But, the 0x12 command is supposed to issue a screen refresh command - which I think should make the screen update seamlessly/quickly . I don't want to have to go through the 25 seconds of screen RESET flash sequence just to update the screen.
For example, using this firmware makes the screen show the current time, so it can be used as a desk clock.
But, every minute the screen updates (because the minute number has changed), and the screen goes through 25 seconds of reset sequence.
Can the screen simply do a refresh (not a reset) ???

Which command does that ?

borodaaaaa, 2024-01-22:
Hey. I also have nebular-350r-n, but cannot hook up pins to it , I have old usb to 323(RJ45) cable. see screenshot.
[image]
 cannot figure how to connect it to the price tag board.

nethomas1968, 2024-01-22:
I don't think that cable will be of any use.
It's got RJ45 on one end, and D-Type on the other. That's not going to be
of any use with this. Sorry.

Regards

Nick

borodaaaaa, 2024-01-22:
Dont look at the connector i cut it so just wires see the pinouts in the pic

nethomas1968, 2024-01-22:
Hi,
I use this setup (see attached pic).
It's got 2 USB to serial converters, and a special pogo pin attachment (the
red thing).

It's maybe hard to see, but one of the USB-Serial adapters is only used for
flashing the new code in to the unit with the SWS, and the CTS pin.

The other USB-Serial is used for VCC and GND, and serial input and output.

Hope that helps.

Regards

Nick Thomas