atc1441/ATC_TLSR_Paper #47: How to connect RTS/RST?

UserXYZ, 2023-12-25:
I am wondering how in the world am I supposed to connect RTS pin from the display to RST pin on UART/serial converter?
I have 3 converters with different chips, and none of them have reset available on the header.

Instructions given here https://github.com/reece15/stellar-L3N-etag say this: The RTS flying wire is connected to the third pin of the CH340G chip (it can also be left unsoldered and manually connected to GND before burning).

But I have no idea which version of CH340 I do have, and their pinout differs between versions.
Plus, pin 3 is RXD on CH340G, while 14 is RST.
Finally, can someone figure out this "manually connected to GND before burning" instruction?

Can anybody explain how am I supposed to connect this pin from the display to any UART converter or eventually to GND itself in order to have the chip put into proper flashing mode?

Thank you.

scriptkitz, 2023-12-25:
RST stands for Reset Pin, which is used to reset the MCU, similar to restarting the MCU. A low level is active. Connecting to RTS is done to make it become low level when UART is connected and set RTS to low. Therefore, let RTS connect to GND, which is the same as low level. You can restart the MCU by programming the UART’s RTS to low level. If your UART does not have an RTS interface, you can manually connect GND to RST to manually restart the MCU. Of course, release the connection to GND after 1 second, then immediately click the burn button. This may require multiple attempts.

UserXYZ, 2024-01-02:
OK, since it was too complicated and tricky to get manual emulation of DTR/RTS from usb-to-serialTTL be done at the correct time, I just ordered two cheap CH340G and FT232 programmers that have exposed all of the signals from the chips.
So by connecting RTS from the converter to the e-paper tag, I reflashed all of them to be sure they are done properly.

I guess that doing short reset puts the microcontroller into programming state, like it does on ESP8266/32, and if you keep it longer, then it goes into full reset.

I am still not sure how did the tags work or even get flashed without pulling the reset pin down before the actual flashing, but they did somehow.

closed 2024-01-02 by UserXYZ