atc1441/ATC_TLSR_Paper #48: how to upload the picture that will stay on screen

UserXYZ, 2024-01-11:
I am trying to find out a way to send a picture to tags using the original atc1441 firmware, notably to BWR154, as the other forks of it only work on larger versions (BWR213, BWR296...) but they have modified the firmware and created a set of web pages/java scripts that allow uploading and keeping the pictures to those larger tags, including dithering and bunch of other cool stuff.

Alas, the smallest one, BWR154, only works with original firmware which accepts an image using hex format (I can convert and create them just fine, uploading them with the usual BLE uploader) but it does not stay on the display - as soon as the time for clock refresh hits, the display is updated and the image disappears, getting replaced by the default clock/battery/temperature combo.

So, is there a way to switch or force original atc1441 firmware to stop this auto update after sending the picture?
Maybe with some specific sequence of hex codes or something?

hnas66, 2024-02-04:
@UserXYZ I cannot help you (yet) with your issue but  I have also a BRW154 
[image]
But the screen is not fully updated, probably du to the firmware I put on it. Do you have the original firmware for me?
thanks in advance for any support you can offer.

HapCoderWei, 2024-06-09:
The epd update at clock time is for there has some code to update it periodly, as a e-clock. If you want to keep your img, you can delete the update code in app.c and recompile a fireware, and upload it into your device.
This code seg is for update img periodly:
```
_attribute_ram_code_ void main_loop(void)
{
    blt_sdk_main_loop();
    handler_time();
// ...

// below is the update code logic
    uint8_t current_minute = (get_time() / 60) % 60;
    if (current_minute != (minute_refresh))
    {
        minute_refresh = current_minute;
        uint8_t current_hour = ((get_time() / 60) / 60) % 24;
        if (current_hour != hour_refresh)
        {
            hour_refresh = current_hour;
            epd_display(get_time(), battery_mv, temperature, 1);
        }
        else
        {
            epd_display(get_time(), battery_mv, temperature, 0);
        }
    }
//...
```
just comment this segment, and your screen should not update by itself, and will contain your picture until you update it again.