atc1441/ATC_TLSR_Paper #50: New layout of Hanshow Nebular 350

teichtaucher, 2024-04-25:
Recently I got an 3.5" hanshow nebular pricetag. Orignally I wanted to reflash it with alternative firmware. But after some cuts into the back of case I found out that they changed the layout - the soldering pads for reflashing are not there as shown in this video:
https://www.youtube.com/watch?v=WRfZJ4xyYwc
Instead it looks like this:
[image]
Besides this other layout the also exchanged the battery type. Now it seems that theres a kind of LiPo instead of the two button cell batteries.
Nevertheless, did anybody of you already manage to reflash this new layout?

MarquezBU, 2024-04-26:
Why destroy the casing so badly?
Simply loosen the plastic cover carefully at the front and everything can be easily removed.

And, your tag is probably about a 350R-N while the video is about the 350Y-N.
But there are no specific differences in functionality or technical specifications between the 350R-N and 350Y-N, except that the 350R-N can display Red, White and Black (BWR) colors.

https://retail-optimiser.de/en/aldi-sued-introduces-esls-from-hanshow/

Check:
https://github.com/atc1441/ATC_TLSR_Paper

atc1441 added support for 3 Color ESL BWR

teichtaucher, 2024-04-26:
You're right, it's a 350R-N. Originally I wanted to open the casing to connect to the pads to reflash the custom firmware (like this https://youtu.be/WRfZJ4xyYwc?t=153). But as I didn't found them at the expected position I opened the casing even more. And then I found out that the layout looks different and I didn't know how to connect the UART flasher to flash the custom firmware. Any idea?

MarquezBU, 2024-04-26:
[image]
[image]
[image]

MarquezBU, 2024-04-26:
Nicht vergessen die Schraube zu lösen... ; - )

teichtaucher, 2024-04-26:
Danke, vielleicht zerlege ich meins auch noch mal ordentlicher. Aber mein Kernproblem löst das allerdings nicht. So wie auf dem zweiten Bild sollte es bei mir auch aussehen - oben links ist die LED, und die Anschlüsse für die Batterie sind auch identisch. Was bei mir aber fehlt sind die Anschlüsse zum Flashen über UART. Die sollten bei mir auf dem Bild am oberen Bildrand liegen.

atc1441, 2024-04-26:
They switch the PCB designs often, even in the video i show 2 versions,

Here its expected that the flashing connector is more in the middle at the thin part(The PCB is very big in your type), here because you got a pouch cell and not CR2450 Batteries

Viel glück

MarquezBU, 2024-04-26:
Du brauchst einen USB to UART Converter.
https://github.com/atc1441/ATC_TLSR_Paper/blob/main/USB_UART_Flashing_connection.jpg

Mach dein Teil mal sauber von vorne auf. Dann Schraube finden und Platine vorsichtig lösen.
Pins sind auf der Rückseite und beschriftet NRST (RTS) TX (TXD) RX (RXD) SWS VCC/GND

teichtaucher, 2024-04-29:
Thanks for the the hints! Finally I opened up the ESL on the front, but in my version it's quite difficult because the pouch cell is glued on the PCB and to the casing. So it's very hard to get the pcb out without breaking the glass of the display. Furthermore there are some pins are underneath the pouch cell:
[image]
But I'm not sure if they can be used for flashing. For me it seems that it's more to check the signals and the connection to the display.
Maybe I will check if I get another PCB revision of Hanshow Nebular 350 or get a Stellar device.

teichtaucher, 2024-04-29:
OK, finally found the Pins foe reflashing:
[image]
It's below the glue sticker at the backside of the PCB. I will now try to flash and Report my status. Hope this helps others.

teichtaucher, 2024-04-29:
Status update:
I could flash the Nebular 3.5" ESL with some restrictions:
-I had to connect also VDD PIN with the USB2UART. Otherwise the LED was not flashing
-TLSR_HS_BWY_35_inverse.bin was working. I also tried TLSR_HS_BWY_35_ALPHA.bin but it shows a black screen with white letters (I would expect this from TLSR_HS_BWY_35_inverse.bin)
-The display shows the content as expected. But the diplay broke during getting it out of the case. So it could not show the content in a clear manner.
- I tried WebBluetooth Image Uploader to upload an image but the ESL could not get found from my PC. On the display of the ESL  "No AP found" is written. What do I need to do to get an connection and to upload an image?

ixy05, 2024-08-25:
Hi @teichtaucher 

From my investigation into this repo and my own experimentation with these devices (Nebular BWR350), you are trying to do two related but different things. The firmware (TLSR_HS_BWY_35_inverse.bin) is from a different repo https://github.com/OpenEPaperLink/OpenEPaperLink. That firmware is set to work with APs that you have to make. This repo works when you upload the .bin file (found here https://github.com/atc1441/ATC_TLSR_Paper/blob/main/Firmware/ATC_Paper.bin).

I may be wrong or this information is not suitable for your experimentations.

teichtaucher, 2024-08-26:
Hi @ixy05 ,
thanks for the hint. Do you know if the bin from @atc1441 's repo (https://github.com/atc1441/ATC_TLSR_Paper/blob/main/Firmware/ATC_Paper.bin) supports control by bluetooth? I stopped further investigation as the step to built up an AP is too much for me. I decided to wait until BLE AP can be used to control Hanshow Nebular 350.