atc1441/ATC_TLSR_Paper #52: Hanshow Nebular 350F-N

unhold, 2024-05-26:
I got my hands on a Hanshow Nebular 350F-N and could flash it via UART with ATC_Paper.bin successfully. However, there is no reaction; the screen does not change and no BLE device appears.

I understand that Nebular devices are not supported, but I have the impression that others had limited success? Any comments are welcome!

[image]

dontrajik, 2024-06-14:
I have a repository with my findings about these tags. I keep it in private but let me know if you would like access to it.

stematon, 2024-06-14:
I can confirm that after flashing thefirmware to a Nebular 350R-N I get a reaction. Approx one half of the screen is populated with the test screen from the FW. Unfortunately the device reboots multiple times per second so I think I need to see if the fw can be modified in a way that the device is more usable.

ixy05, 2024-06-23:
> I have a repository with my findings about these tags. I keep it in private but let me know if you would like access to it.

Hi dontrajik,

Would you be able to make this public please? I'd love to find out more about them.

Thanks!

NoviceArtisan, 2024-07-15:
> I have a repository with my findings about these tags. I keep it in private but let me know if you would like access to it.

Could I have a look too please? Cheers

Felix-Walton, 2024-07-29:
> I have a repository with my findings about these tags. I keep it in private but let me know if you would like access to it.

Would love to have a look too, thanks!

dontrajik, 2024-08-05:
@NoviceArtisan @Felix-Walton

Made the repo public:

https://github.com/dontrajik/Nebular

tobiasdroessler, 2024-12-02:
@dontrajik could you explain what's inside?

It is a fork of this repo?

valtoree, 2025-05-24:
is this device done?
because on recent video of Aaron https://www.youtube.com/watch?v=9oKWkHGI-Yk
at 7m he shows what appears to be one. i have one and am dying to make us of it.
i see on his ATC_BLE_OEPL Uploader website there are different 350 HS BW etc but I don't know if it's in there. can someone investigate pls?
https://atc1441.github.io/ATC_BLE_OEPL_Image_Upload.html

valtoree, 2025-05-24:
@atc1441 tx so much for reply 🙏
ok so I guess I wait some more 😪
keep up the great work, your amazing!

atc1441, 2025-05-24:
@mukowman exactly that :)

valtoree, 2025-05-26:
ty @mukowman & @atc1441 for replies 🙏
i joined the discord and after some research was able to successfully flash oepl to both my 350f-n and 350r-n ♥️🎉 now to get more eink price tags!! 😁