atc1441/ATC_TLSR_Paper #54: TLSR8250 chip compatibility

sgilliar37, 2024-09-03:
I just came across a similar device:

[image]

and I'm looking to see how I can use this device to send data to the display.  It's a TLSR8250 chip and it has a 2.13in BW e-ink display on the front.  If you have any info on whether this will work with this project, I would greatly appreciate it.

Steve

tpruszin, 2025-01-16:
@sgilliar37 Did you ever find a solution to this? or have the stock firmware for this tag?

seb633, 2026-11-10:
Have you soIved this problem？I came to the same issue.

sgilliar37, 2026-11-11:
No, still unresolved.

seb633, 2026-11-11:
try similar size screen drive?

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2026年11月11日 21:21 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [atc1441/ATC_TLSR_Paper] TLSR8250 chip compatibility (Issue #54) |
sgilliar37 left a comment (atc1441/ATC_TLSR_Paper#54)

No, still unresolved.

sgilliar37, 2026-11-11:
Working on this this weekend.  Hope to have something to show after.