atc1441/ATC_TLSR_Paper #58: M3N display issues

gabimal, 2024-12-22:
Hello everyone!
After flashing the firmware on this ESL the display looks washed out and the digits seems to be incomplete. I used UART_Flasher.html 
Any ideea why? 

[image]

atc1441, 2024-12-22:
Hey,  that type is currently not supported

gabimal, 2024-12-22:
But it is the same as the above one that works ok.

atc1441, 2024-12-22:
While they look the same the internals differ on some so it can not be guaranteed

gabimal, 2024-12-22:
The one that works is in the right side of this picture:

[image]

Strange...

atc1441, 2024-12-22:
It is luck what internals you get :) 
And that specific one is an unlucky one^^

gabimal, 2024-12-22:
[image]

gabimal, 2024-12-22:
Aaron are you still working/developing this ESL project? It would be nice to have some sort of BT node that updates the ESLs based on their MAC address, just like OpenEPaperLink. 
Is it douable?

atc1441, 2024-12-22:
This codebase is not developed any further, the OpenEpaperLink Project is now the way to go :)

atc1441, 2024-12-22:
https://youtu.be/e1vod_OqKys

gabimal, 2024-12-22:
So there is a new FW that can be pushed on these ESLs to be integrated in OpenEpaperLink project?

atc1441, 2024-12-22:
Its all in the video :)
That specific display is most likely not included right now as well as I do not own it to adapt it.

gabimal, 2024-12-22:
Thank you so much for your help Aaron! 
I can do trial and error with fw from here, just tell me how can I help. I do have the M3N and XL3N for which I created a post here: https://github.com/atc1441/ATC_TLSR_Paper/issues/11

I wish you a Merry Christmas and a Happy new Year!

atc1441, 2024-12-22:
Thank you as well!
I will only add new displays right now that I can fully test(Have^^) At a later stage the firmware will include an user editable display adder at runtime. But thats taking a bit.

And Merry Christmas to you too!

gabimal, 2024-12-22:
I can send you the displays that I have for you to fully test and integrate them in this awesome project. Are you ok with that?

atc1441, 2024-12-22:
Please write a mail to eink@43u.de

closed 2025-01-05 by gabimal