atc1441/ATC_TLSR_Paper #59: May I ask how to modify the screen resolution？

zhaoqi90, 2025-01-12:
”XM3N@---E31A “ This model has a resolution of 296 * 152, I'm not sure where to change it，https://github.com/reece15/stellar-L3N-etag 
[image]

Fred2204, 2025-01-12:
Do you want to get rid of the black and white noise at the bottom?

zhaoqi90, 2025-01-13:
> 你想摆脱底部的黑白杂色吗？

Of course, do you have any good suggestions？

Fred2204, 2025-01-13:
To me it looks like that the width or height variable is not correct. Which EPD model number is detected when you run the code?

zhaoqi90, 2025-01-13:
The default is “EPD296”, I tried to modify item ”0X44“
 ”// Set RAM X- Address Start/End
    EPD_WriteCmd(0x44);
    EPD_WriteData(0x00);
    EPD_WriteData(0x12);“
After modification, the screen is blurred，Do not display

Fred2204, 2025-01-13:
I thought that the code has only models 154, 213 or 350. Which one are you using?
And I thought resolution setting is command 0x61.

Fred2204, 2025-01-13:
I think your height variable is set to 128 and you need to change it to 152.

zhaoqi90, 2025-01-13:
https://github.com/reece15/stellar-L3N-etag 
This is the EPD296 version, according to its description, it needs to be changed to 0x44

zhaoqi90, 2025-01-13:
This EDP is 296 * 128, I just need to set 128>152.

zhaoqi90, 2025-01-13:
This EDP is 296 * 128, I just need to set 128>152.
“https://javabin.cn/2022/epaper.html”