atc1441/ATC_TLSR_Paper #60: Nebular 350R-N difference between SSD and ALPHA

elvis-hai-le, 2025-04-04:
Hi all, I'm currently looking into getting the Nebular 350R-N working, and I noticed that there are working firmwares for the tags in the [openepaperlink](https://github.com/OpenEPaperLink/OpenEPaperLink/tree/master/binaries/Tag) repository. 

This question is mainly directed at Aaron as he was able to make full use of the display height despite mentioning the limitations of the microcontrollers themselves. I've gotten as far as making the display the height of 168px but can't quite seem to get it to reach 184px. I had thought perhaps it was by using partial updates on the screen but when the correct firmware is flashed, the display updates in one go so I'm not sure if this is the right direction but I'll most likely try it next. I also know that the one bit display stuff is only used on a portion of the screen as I was able to replicate results similar to the initial image of openepaperlink on the Alpha version. 

Aaron, I hope if you have the time, could you please let me know what you did to get the SSD version to use the whole screen and able to display a white background as well? I'm in awe of the work you've done so far and would love to be able to contribute if possible. 

Please don't hesitate to post any questions or ideas you guys have as well!

Cheers,
Elvis

[image]

elvis-hai-le, 2025-04-04:
I also sent an email to your Youtube email Aaron, sorry to bother you 😅

elvis-hai-le, 2025-04-04:
As a follow-up you can disregard the full height factor, I was able to successfully use the full height by sacrificing the width to about 200px. I'll update as I make more discoveries [image]

atc1441, 2025-04-04:
Hey, you can you the ATC_BLE_OEPL firmware with them, there are fully supported there: https://youtu.be/e1vod_OqKys

elvis-hai-le, 2025-04-04:
Oh cheers, I'll give that one a go, I didn't realise that was an option as well. Thanks Aaron!

elvis-hai-le, 2025-04-07:
I've tried that firmware and it works for the alpha but the ssd seems to just disconnect after attempting to set any display type. When I send an image it confirms the image is displayed but there is no update to the display, I've tried reflashing multiple times but seems like its just not happy with the firmware. I tried BWY SSD and BWR SSD 350s for the aforementioned attempts.

elvis-hai-le, 2025-04-11:
All good, everything is working now, flashed the new firmware and was able to upload images. Cheers

StefCoders, 2025-04-21:
Any way to flash them through NFC, don't want to break mine. It might be impossible though.

kioskaskdn, 2026-09-23:
> Any way to flash them through NFC, don't want to break mine. It might be impossible though.

you dont need to break them - i used my wifes hair dryer, heated the fascia up well, then slightly bent the tag and front panel popped out - no breaks anywhere.