atc1441/ATC_TLSR_Paper #61: Hanshow Nebular 350R-N

StefCoders, 2025-04-21:
Hey everyone, great repo you have here.

I recently got one of these given to me and I am unsure how to put something else on them.

I wouldn't like to destroy them but also want to hack them.

Here are some pictures.

[image][image][image]

mukowman, 2025-05-21:
Essentially you want to use a soldering iron to melt a section on the case to expose the contacts.
Then use a UART flasher and pogo pins to flash like https://www.amazon.com.au/dp/B09TW2QX21?ref=ppx_yo2ov_dt_b_fed_asin_title (preferable one with reset).

These are the pins at the top

[image]

This is the melting location

[image]

Use this page to flash
https://atc1441.github.io/ATC_TLSR_Paper_UART_Flasher.html

Use this firmware 
https://github.com/atc1441/atc1441.github.io/blob/master/ATC_BLE_OEPL.bin

Then manage with this page over BLE
https://atc1441.github.io/ATC_BLE_OEPL_Image_Upload.html

nethomas1968, 2025-06-19:
You will struggle to get any use out of this EPD device even if you can program it.
The code in this repo doesn't work very well with the screen size. There is still some work to do to get it to work properly.
My experience wasn't very successful.
I used a USB-Serial method to program the device, which works fine. But, the on screen display isn't useful.
Good luck though.

atc1441, 2025-06-19:
The latest and simplest so far flashing method can be found here: https://youtu.be/9oKWkHGI-Yk
The project in this repo is very old and not updated anymore, the newer firmware has support for much more display types

atc1441, 2025-06-19:
> Where is the newer firmware? Is that a different repo ?

It will be automatically included when flashed as shown in the video :)

ashimokawa, 2025-07-03:
@atc1441 

Is publishing the source code planned? Or will it stay closed?

schools-dev, 2026-05-02:
uh I'm 13 I don't have access to soldering.. am I cooked